ros-misc-utilities/ffmpeg_image_transport #3: Error decoding ffmpeg packets.

akssri-sony, 2023-04-26:
I run into the following ffmpeg decoder error on the sub-side under this setup,
- The publisher is a standard image publisher (intel-realsense), running in a ROS2 env with ffmpeg-transport compiled/installed.
- Both the publisher/subscriber are running humble and don't have GPUs.  
- Subscriber is a image_transport republisher launched with standard republish command (as given in README without params).

```
....
[WARN] [1682494335.377477553] [FFMPEGDecoder]: send_packet failed for pts: 202
[h264 @ 0x55b21f7f68c0] Invalid setup for format cuda: missing configuration.
Failed to get HW surface format.
[h264 @ 0x55b21f7f68c0] decode_slice_header error
[h264 @ 0x55b21f7f68c0] no frame!
[WARN] [1682494335.444379691] [FFMPEGDecoder]: send_packet failed for pts: 203
[h264 @ 0x55b21f7f68c0] Invalid setup for format cuda: missing configuration.
Failed to get HW surface format.
[h264 @ 0x55b21f7f68c0] decode_slice_header error
[h264 @ 0x55b21f7f68c0] no frame!
[WARN] [1682494335.510952361] [FFMPEGDecoder]: send_packet failed for pts: 204
^C[INFO] [1682494335.541615526] [rclcpp]: signal_handler(signum=2)
...
``` 

Would be grateful if you have some insight into what's going wrong or how this could solved. Thanks!

berndpfrommer, 2023-04-26:
Could you please post the output (first few lines) of  'ros2 echo /your_image_topic'? I'd like to see what encoding is being used by the encoder?

The problem is apparently that the decoder tries to use some cuda decoder. Not sure why. Could you please also post the line with which you launch the republisher (decoder)?

The command for starting the republisher in the README almost certainly will have to be modified to avoid using the accelerated decoder. Maybe something like this:

`ros2 run image_transport republish ffmpeg in/ffmpeg:=image_raw/ffmpeg raw out:=image_raw/uncompressed --ros-args -p "ffmpeg_image_transport.map.hevc_nvenc:=libx264"`

akssri-sony, 2023-04-27:
This is what I get when I echo the topic with `ros2 topic echo /camera/color/image_rect_raw/ffmpeg`
```
header:
  stamp:
    sec: 1682568513
    nanosec: 971701760
  frame_id: camera_color_optical_frame
width: 848
height: 480
encoding: libx264
pts: 1394
flags: 0
is_bigendian: false
data:
- 0
- 0
- 0
- 1
- 65
- 155
- 211
- 120
- 67
- 252
- 153
- 76
- 0
- 130
- 31
- 130
- 15
- 106
- 70
- 132
- 92
- 54
- 11
- 5
- 182
- 224
- 235
- 183
- 147
- 56
- 48
- 149
- 103
-
```

The republisher was initially run with `ros2 run image_transport republish ffmpeg in/ffmpeg:=/camera/color/image_rect_raw/ffmpeg raw out:=/camera/color/image_rect_raw/uncompressed`. 

I tried setting all the mappings with
```
ros2 run image_transport republish ffmpeg in/ffmpeg:=/camera/color/image_rect_raw/ffmpeg raw out:=/camera/color/image_rect_raw/uncompressed --ros-args -p "ffmpeg_image_transport.map.h264_nvmpi:=libx264"  --ros-args -p "ffmpeg_image_transport.map.h264_nvenc:=libx264"  --ros-args -p "ffmpeg_image_transport.map.hevc_nvenc:=libx264" --ros-args -p "ffmpeg_image_transport.map.libx264:=libx264"
```

Only the last one had any effect (since the stream is encoded with libx264). Now I get,
```
root@garuda:/home/ros/humble# ros2 run image_transport republish ffmpeg in/ffmpeg:=/camera/color/image_rect_raw/ffmpeg raw out:=/camera/color/image_rect_raw/uncompressed --ros-args -p "ffmpeg_image_transport.map.h264_nvmpi:=libx264"  --ros-args -p "ffmpeg_image_transport.map.h264_nvenc:=libx264"  --ros-args -p "ffmpeg_image_transport.map.hevc_nvenc:=libx264" --ros-args -p "ffmpeg_image_transport.map.libx264:=libx264"
[WARN] [1682571049.574895085] [rcl]: Found remap rule 'in/ffmpeg:=/camera/color/image_rect_raw/ffmpeg'. This syntax is deprecated. Use '--ros-args --remap in/ffmpeg:=/camera/color/image_rect_raw/ffmpeg' instead.
[WARN] [1682571049.574909802] [rcl]: Found remap rule 'out:=/camera/color/image_rect_raw/uncompressed'. This syntax is deprecated. Use '--ros-args --remap out:=/camera/color/image_rect_raw/uncompressed' instead.
[ERROR] [1682571049.637535289] [image_republisher]: SubscriberPlugin::subscribeImpl with five arguments has not been overridden
[ERROR] [1682571050.100268191] [FFMPEGDecoder]: cannot open decoder libx264
[ERROR] [1682571050.100311730] [FFMPEGDecoder]: cannot open decoder libx264
[ERROR] [1682571050.100349746] [FFMPEGSubscriber]: cannot initialize decoder!
[ERROR] [1682571051.100484505] [FFMPEGDecoder]: cannot open decoder libx264
[ERROR] [1682571051.100518457] [FFMPEGDecoder]: cannot open decoder libx264
[ERROR] [1682571051.100523935] [FFMPEGSubscriber]: cannot initialize decoder!

```
Some digging on the internet led me to something that says that ffmpeg needs to be recompiled with spec. flags; not sure if this is the issue.

akssri-sony, 2023-04-27:
Looks like `get_hw_type` in ffmpeg_decoder.cpp always returns the `AV_HWDEVICE_TYPE_CUDA` (=2), regardless of whether there's a cuda device or not. Hacking this to take the non-Cuda code path leads to other errors: the decoded image is all messed up. Thanks for the help!

Input:
[image]

Output:
[image]

berndpfrommer, 2023-04-27:
What happens if you run the republisher like this:

`ros2 run image_transport republish ffmpeg in/ffmpeg:=image_raw/ffmpeg raw out:=image_raw/uncompressed --ros-args -p "ffmpeg_image_transport.map.libx264:=h264"
`
If you use straight ffmpeg (the executable) from the command line, can you encode/decode with libx264? If that works, then the ROS2 transport plugins should also work.

berndpfrommer, 2023-04-27:
Oh, and your image resolution is really odd. The line size of 848 is not even a multiple of 32. Other users have reported weirdly decoded images in such situations.
I'm surprised you didn't get a warning on the encoder side:
```
if ((width % 32) != 0) {
      RCLCPP_WARN(logger_, "horiz res must be multiple of 32!");
    }

```

akssri-sony, 2023-04-28:
If I run the republisher with,
```
ros2 run image_transport republish ffmpeg in/ffmpeg:=image_raw/ffmpeg raw out:=image_raw/uncompressed --ros-args -p "ffmpeg_image_transport.map.libx264:=h264" 
```
I end up with the same error as in the first comment.

On the resolution side, I did get the warning, but didn't realize this would be the effect. Is there an easy fix for this. I guess padding would work, but is there a better way on the libav side ?

berndpfrommer, 2023-04-28:
I really don't know how to fix this on the libav side. But does standalone
ffmpeg work correctly for decoding h264 packets?

berndpfrommer, 2023-05-21:
Fixed by PR #4

closed 2023-05-21 by berndpfrommer