ros-misc-utilities/ffmpeg_image_transport #13: h264_vaapi

fferroni, 2023-09-04:
Hi,
Have you ever tried using `h264_vaapi` as an encoder? It doesn't seem to accept `preset` options.
https://trac.ffmpeg.org/wiki/Hardware/VAAPI#Mappingoptionsfromlibx264

berndpfrommer, 2023-09-04:
I have never tried vaapi. What error are you getting? From looking at the code, it should continue even if setting the preset returns an error. What's not working?

fferroni, 2023-09-06:
It complains that it does not accept `preset` as a parameter.

berndpfrommer, 2023-09-06:
The encoding looks quite different from what worked for nvenc. Here is an example of vaapi encoding:
https://github.com/FFmpeg/FFmpeg/blob/master/doc/examples/vaapi_encode.c

berndpfrommer, 2023-09-20:
I added support for VAAPI. Tested on AMD laptop.

closed 2023-09-29 by berndpfrommer