ros-misc-utilities/ffmpeg_image_transport #20: Pixel format '(null)' not supported

dwffls, 2023-11-16:
When using the h264_vaapi encoder AVHWFrameContext gives the following error: 
```
[AVHWFramesContext @ 0x7eff081a2100] The hardware pixel format '(null)' is not supported by the device type 'VAAPI'
[ERROR] [1700131267.578674463] [FFMPEGEncoder]: failed to initialize VAAPI frame context: Function not implemented
[ERROR] [1700131267.579830043] [FFMPEGPublisher]: cannot initialize encoder!
```

When using the default libx264 encoder everything works as expected. 

VAAPI is correlty installed on my machine with the following vainfo output:
```
libva info: VA-API version 1.7.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_7
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.7 (libva 2.6.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 20.1.1 ()
vainfo: Supported profile and entrypoints
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileNone                   : VAEntrypointStats
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSliceLP
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSliceLP
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointEncSliceLP
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile1            : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
      VAProfileVP9Profile3            : VAEntrypointVLD
      VAProfileHEVCMain12             : VAEntrypointVLD
      VAProfileHEVCMain422_10         : VAEntrypointVLD
      VAProfileHEVCMain422_12         : VAEntrypointVLD
      VAProfileHEVCMain444            : VAEntrypointVLD
      VAProfileHEVCMain444            : VAEntrypointEncSliceLP
      VAProfileHEVCMain444_10         : VAEntrypointVLD
      VAProfileHEVCMain444_10         : VAEntrypointEncSliceLP
      VAProfileHEVCMain444_12         : VAEntrypointVLD
      VAProfileHEVCSccMain            : VAEntrypointVLD
      VAProfileHEVCSccMain10          : VAEntrypointVLD
      VAProfileHEVCSccMain444         : VAEntrypointVLD
```

dwffls, 2023-11-16:
Setting ffmpeg_image_transport.pixel_format:=nv12 did not result in a different result

dwffls, 2023-11-16:
The ffmpeg example runs without any problem. 
Next to this, ive already tried to looking at/ changing the code snippet you provided. However without any luck.
It appears as though net usesHardwareFrames_ bool is never set correctly. 

I'll let you know if i find out anything more

berndpfrommer, 2023-11-29:
@dwffls have you made any progress on this issue? If usesHardwareFrames_ is false, then that means something is going wrong in [find_hw_config()](https://github.com/ros-misc-utilities/ffmpeg_image_transport/blob/74c7b9bc5c26f51a255e3731d3d546dcf4bf6394/src/utils.cpp#L61). Can you put some debug printing in there to find out why usesHWFrames is not set correctly? 
It could be that either the device type does not match or config->methods does not have the right flags set.

berndpfrommer, 2023-12-06:
closing for lack of activity. Not clear what the problem is.

closed 2023-12-06 by berndpfrommer