ros-misc-utilities/ffmpeg_image_transport #24: hardware conversion to maximize performance

buckleytoby, 2023-12-23:
I've created a branch entitled 'cuda_conversion' on my fork: [link here](https://github.com/NU-Haptics-Lab/ffmpeg_image_transport/tree/cuda_conversion).

This uses a cuda kernel to convert the software format to the correct hardware format the GPU expects. 

Performance improvement is as follows:
<google-sheets-html-origin>
Start | End | Delay (ms) | Method
-- | -- | -- | --
encodeImage() | /cam0/image_raw/ffmpeg | 12.6 | swscale
encodeImage() | /cam0/image_raw/ffmpeg | 3.2 | Cuda Conversion

berndpfrommer, 2023-12-23:
Nice work and pretty impressive performance gains!
It's just that I don't quite know what to do with this issue
You may be a aware that I'm trying to release ffmpeg_image_transport as a ros distro package (e.g. installable with apt under ubuntu). So far I have been able to avoid an outright cuda dependency as there may be platforms that don't support it or users who wouldn't want to install it.  This means outright merging your changes is not possible. So your branch would have to be packaged into a separate package, e.g. ffmpeg-image-transport-cuda. That could be done if you want to go this route.
[Minor wrinkle: part of the code you contributed is licensed under GPL2 and I'm not a particular fan of GPL licenses. I don't think it's compatible with Apache 2. So if you can find those routines with a more permissive license that would be great.]

Another alternative: do you want to move your branch into this repo? Users would have to build from source, but it may make it easier to find.

buckleytoby, 2023-12-23:
Thanks! Learning how to interface cuda with libavcodec was a real journey.
I also didn't know where it would best fit in, and I defer to your judgement. Whatever works!

berndpfrommer, 2023-12-24:
I gave you write access. Can you try to push your branch to the repo?

buckleytoby, 2024-02-01:
I pushed to the branch https://github.com/ros-misc-utilities/ffmpeg_image_transport/tree/cuda_conversion

closed 2024-02-01 by buckleytoby