ros-misc-utilities/ffmpeg_image_transport #27: Error decoding libx264 on Jetson Orin - Cannot load libnvcuvid.so.1

jmm-slamcore, 2024-02-07:
I installed the `ros-iron-ffmpeg-image-transport` package via apt on a Jetson Orin NX (Publisher) and a Jetson Orin AGX (Subscriber) and get an error when trying to decode the ffmpeg topic without explicitly using hardware acceleration:

Publisher - Orin NX running Jetpack 6 (Ubuntu 22) works ok:

```
ros2 run image_transport republish raw in:=/amr/infrared_0/image_raw ffmpeg out/ffmpeg:=/amr/infrared_0/image_raw/ffmpeg
```

```
[WARN] [1707337736.418144200] [rcl]: Found remap rule 'in:=/amr/infrared_0/image_raw'. This syntax is deprecated. Use '--ros-args --remap in:=/amr/infrared_0/image_raw' instead.
[WARN] [1707337736.418209354] [rcl]: Found remap rule 'out/ffmpeg:=/amr/infrared_0/image_raw/ffmpeg'. This syntax is deprecated. Use '--ros-args --remap out/ffmpeg:=/amr/infrared_0/image_raw/ffmpeg' instead.
[INFO] [1707337736.674866902] [FFMPEGEncoder]: enc: libx264 prof:  preset: 
[INFO] [1707337736.674984091] [FFMPEGEncoder]: qmax: 10 bitrate: 8242880 gop: 15
[WARN] [1707337737.385677712] [FFMPEGEncoder]: horiz res must be multiple of 32!
[libx264 @ 0xaaab16e81130] using cpu capabilities: ARMv8 NEON
[libx264 @ 0xaaab16e81130] profile High, level 3.2, 4:2:0, 8-bit
[INFO] [1707337737.390063329] [FFMPEGEncoder]: opened codec: libx264
```

Subscriber - Orin AGX running Jetpack 6 (Ubuntu 22) throws an error:

```
ros2 run image_transport republish ffmpeg in/ffmpeg:=/amr/infrared_0/image_raw/ffmpeg raw out:=/amr/infrared_0/image_raw/ffmpeg_uncompressed
```

```
[WARN] [1707338504.776305805] [rcl]: Found remap rule 'in/ffmpeg:=/amr/infrared_0/image_raw/ffmpeg'. This syntax is deprecated. Use '--ros-args --remap in/ffmpeg:=/amr/infrared_0/image_raw/ffmpeg' instead.
[WARN] [1707338504.776384174] [rcl]: Found remap rule 'out:=/amr/infrared_0/image_raw/ffmpeg_uncompressed'. This syntax is deprecated. Use '--ros-args --remap out:=/amr/infrared_0/image_raw/ffmpeg_uncompressed' instead.
[INFO] [1707338506.496957587] [FFMPEGDecoder]: decoding with h264
[h264 @ 0xaaaaf0517390] Cannot load libnvcuvid.so.1
[h264 @ 0xaaaaf0517390] Failed loading nvcuvid.
[h264 @ 0xaaaaf0517390] Failed setup for format cuda: hwaccel initialisation returned error.
Failed to get HW surface format.
[h264 @ 0xaaaaf0517390] decode_slice_header error
[h264 @ 0xaaaaf0517390] no frame!
[WARN] [1707338506.499019759] [FFMPEGDecoder]: send_packet failed for pts: 11503
```

I have also tried 

```
ros2 run image_transport republish ffmpeg in/ffmpeg:=/amr/infrared_0/image_raw/ffmpeg raw out:=/amr/infrared_0/image_raw/ffmpeg_uncompressed --ros-args -p "ffmpeg_image_transport.map.hevc_nvenc:=libx264"
```
and
```
ros2 run image_transport republish ffmpeg in/ffmpeg:=/amr/infrared_0/image_raw/ffmpeg raw out:=/amr/infrared_0/image_raw/ffmpeg_uncompressed --ros-args -p "ffmpeg_image_transport.map.libx264:=h264"
```
as suggested in another issue but I get the same. I think hardware accelerated ffmpeg is not fully supported on jetson out of the box but I should be able to run this without hardware acceleration?

berndpfrommer, 2024-02-08:
Is the libnvcuvid installed?
https://www.reddit.com/r/jellyfin/comments/10cdxhw/comment/j4fgsbx

berndpfrommer, 2024-02-20:
If there is no more interest I will close this issue.

jmm-slamcore, 2024-02-20:
Wasn't able to resolve the issue with hardware acceleration and didn't have time to look into it further so I have opted to resize the image with image_proc instead of compressing it for now. Feel free to close this.

berndpfrommer, 2024-02-20:
Seems like the Jetson Orin still does not support nvenc. See issue #28 
There is a way to compile a custom version of libffmpeg with  NVMPI support, then link ffmpeg_image_transport against that library. Not for the faint hearted, but I got it working for encoding a single image stream (stereo fails due to #stream limitations).

Closing this issue.

closed 2024-02-20 by berndpfrommer