ros-misc-utilities/ffmpeg_image_transport #29: How do I change the library?

seherrr, 2024-03-15:
Hi, I have a question. I am using orangepi5b. I built the rkmpp ecoder for ffmpeg using the https://github.com/nyanmisaka/ffmpeg-rockchip/tree/master repo.

In this way ./ffmpeg -f lavfi -i testsrc2=s=1920x1080,format=nv12 -c:v hevc_rkmpp -qp_init 26 -profile:v main -level 4.1 -g:v 100 -vframes 5000 -y /tmp/tmp. I can do mp4 encoding successfully.

I could not manage to use the ffmpeg libraries() that I built in the ffmpeg_image_transport repo.

For example, libacvodec/avcodec.h uses the file path as usr/include/aarch64/-linux-gnu/libavcodec/avcodec.h. How can I change this directory? How do I make it use the /root/dev/ffmpeg/libavcodec/avcodec.h directory?

I set DFFMPEG_PKGCONFIG and LD_LIBRARY_PATH environment variables with the colcon mentioned in the readme, but I did not get the result I wanted.

berndpfrommer, 2024-03-15:
When you are using this line to build:
```
colcon build --symlink-install --cmake-args --no-warn-unused-cli -DFFMPEG_PKGCONFIG=/home/foo/ffmpeg/build/lib/pkgconfig -DCMAKE_BUILD_TYPE=RelWithDebInfo
```
it shoud be picking up the library in ``/home/foo/ffmpeg/build/lib``. Apparently it's not doing that.
1) Double and triple check you have no typos in your path name to the custom build directory. Any mistake there will mean the build process silently picks up the system-wide installed header files, which is not what you want. I always do ``ls -l <path_name>``, and cut-and-pasting the path name so I catch any mistakes.
2) Run ``VERBOSE=1 colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DFFMPEG_PKGCONFIG=/home/foo/ffmpeg/build/lib/pkgconfig --event-handlers console_direct+``. This will give you a more verbose output of the build process. How does the search path (-I (include)) look on the compiler g++ line? Does it point to your custom library?
3) Run your compile as ``strace -f <here the entire colcon compile command> >& /tmp/foo.txt``. This will produce a huge amount of output but it will show you what system calls the build process makes. Is it ever looking in the right places, i.e. is it using your custom path? Grep for the pkgconfig file that you expect it to open, you should be able to find it, or see where it grabs the wrong one. Maybe you get some inspiration from there about what's going wrong.

seherrr, 2024-03-19:
My problem is that I added this apart from what you added. The shortest way to do so is to add --enable-shared at ./configure options

./configure --prefix=/usr --enable-gpl --enable-version3 --enable-libdrm --enable-rkmpp --enable-rkrga --enable-shared

closed 2024-03-19 by seherrr

berndpfrommer, 2024-03-19:
Thank you for clarifying. Could you please comment on how the issue with not finding the include files was resolved? Just adding the --enable-shared hardly did the trick.
I do see however that you have as prefix /usr, which would mean that upon installation you will overwrite the system-installed ffmpeg. That may not be what you wanted.