ros-misc-utilities/ffmpeg_image_transport #30: Configurable CRF?

tonynajjar, 2024-05-07:
CRF is the first configuration point in the [website](https://trac.ffmpeg.org/wiki/Encode/H.264#a1.ChooseaCRFvalue) but doesn't seem configurable. Any good reason or just not implemented?

berndpfrommer, 2024-05-07:
I wasn't aware of CRF, if that qualifies as a good reason. I don't see any CRF setting in the [AVCodecContext struct](https://ffmpeg.org/doxygen/trunk/structAVCodecContext.html). Way back when I wrote the first version of this I got to where I wanted by tuning "qmax". Be happy to see a PR to add a knob for CRF.

tonynajjar, 2024-05-07:
I see it [here](https://ffmpeg.org/doxygen/0.6/structAVCodecContext.html#e00c86e8f984777d11eb2fb6bea0dfb7) which is version 0.6. Your points to "trunk", not sure what that is

tonynajjar, 2024-05-07:
I searched for a bit and really don't see an obvious way to do this. I guess I'll content myself with other knobs for the time being

shtern, 2024-11-04:
Fixed in #34