ros-misc-utilities/ffmpeg_image_transport #32: set stream bitrate

radisc, 2024-08-29:
Hello everyone,

I'm using ffmpeg image transport with v4l2_camera_node, with a 640x380@30fps video stream on a raspberrypi 5.
I can set the encoder to use, the profile , the preset, etc. But there i can't set the bit_rate: whatever i set, the stream is going to take ~ 3MB/s ( from gnome-system-monitor read on the receiver).

This is cam.launch.py
```
     ld.add_action(
        Node(
            package='v4l2_camera',
            executable='v4l2_camera_node',
            output='screen',
            name="v4l2_camera",
            parameters=[
                params_path,
                {
                    'ffmpeg_image_transport.bit_rate': 2000000,
                    'ffmpeg_image_transport.encoding': 'libx264',
                    # 'ffmpeg_image_transport.encoding': 'libx264',
                    # 'ffmpeg_image_transport.encoding': 'hevc_nvenc',
                    'ffmpeg_image_transport.tune': 'zerolatency',
                    'ffmpeg_image_transport.profile': 'baseline',
                    'ffmpeg_image_transport.preset': 'superfast',
                    'ffmpeg_image_transport.gop': 15,
                },
            ],
        )
    )
```

And this is the stdout ( minus v4l2camera stuff )
```
[v4l2_camera_node-1] [INFO] [1724941076.329727219] [FFMPEGEncoder]: enc: libx264 prof: baseline preset: superfast
[v4l2_camera_node-1] [INFO] [1724941076.329840089] [FFMPEGEncoder]: qmax: 10 bitrate: 2000000 gop: 15
...
[v4l2_camera_node-1] [libx264 @ 0x7ffef0005120] using cpu capabilities: ARMv8 NEON
[v4l2_camera_node-1] [libx264 @ 0x7ffef0005120] profile Constrained Baseline, level 3.2, 4:2:0, 8-bit
[v4l2_camera_node-1] [INFO] [1724941076.829477650] [FFMPEGEncoder]: opened codec: libx264
```
With this preset the quality and the latency are perfect for my application, but i would be happy if i could lower the bandwidth a little.
Thanks

berndpfrommer, 2024-08-29:
Sorry I can't be of much help there. I implemented the bit_rate parameter, but never really looked to carefully into how it interacts with other parameters. In fact I set bit_rate way higher than necessary and then just played with the other parameters until 
 the video looked good.
In your case maybe your qmax of 10 is too small, i.e. you are requesting too high quality there, and then the encoder violates the bit_rate constraint.
If you want to play around in more detail you can hack the encoder directly, around [this line](https://github.com/ros-misc-utilities/ffmpeg_image_transport/blob/09c09cbdc8d662e65a537014f7be09c1d82960b5/src/ffmpeg_encoder.cpp#L204), setting (or not setting the) parameters until you get the encoder to honor the bit_rate parameter.

radisc, 2024-09-02:
Thanks, changing qmax already unblocked the bitrate

berndpfrommer, 2024-09-12:
ok, closing this as solved

closed 2024-09-12 by berndpfrommer