ros-misc-utilities/ffmpeg_image_transport #43: Cannot decode AV1

christianrauch, 2025-05-31:
I am trying to encode and decode `av1`. `ffmpeg -codecs` shows me that this is possible with a variety of encoders and decoders:
```
 DEV.L. av1                  Alliance for Open Media AV1 (decoders: libdav1d libaom-av1 av1 av1_cuvid av1_qsv) (encoders: libaom-av1 librav1e libsvtav1 av1_nvenc av1_qsv av1_vaapi)
```

For this, I am running [`v4l2_camera_node`](https://index.ros.org/p/v4l2_camera/) with `ffmpeg.encoding:=librav1e`:
```sh
ros2 run v4l2_camera v4l2_camera_node --ros-args -p video_device:=/dev/video2 -p image_raw.ffmpeg.encoding:=librav1e
```
and I can see the `FFMPEGPacket` messages on the transport topic:
```yaml
header:
  stamp:
    sec: 1748708693
    nanosec: 760680343
  frame_id: camera
width: 640
height: 480
encoding: librav1e
pts: 3
flags: 0
is_bigendian: false
data: '<sequence type: uint8, length: 28158>'
```

To view the images, I am using rqt and per https://github.com/ros-misc-utilities/ffmpeg_image_transport/?tab=readme-ov-file#subscriber-viewer, I am trying to set the decoder. But when I list the parameters for the rqt node (`ros2 param list /rqt_gui_cpp_node_644409`), I only see encoders:
```
  image_raw.ffmpeg.map.h264_nvenc
  image_raw.ffmpeg.map.h264_nvmpi
  image_raw.ffmpeg.map.h264_vaapi
  image_raw.ffmpeg.map.hevc_nvenc
  image_raw.ffmpeg.map.librav1e
  image_raw.ffmpeg.map.libx264
  image_raw.ffmpeg.map.measure_performance
```
and it won't let me set another parameter:
```sh
$ ros2 param set /rqt_gui_cpp_node_644409 image_raw.ffmpeg.map.libdav1d av1
Setting parameter failed: parameter 'image_raw.ffmpeg.map.libdav1d' cannot be set because it was not declared
```

First, I am wondering why the `FFMPEGPacket` message is using `librav1e` instead of `av1` for `encoding`. Does it matter for the receiver which encoder was used? Shouldn't it be enough to get the raw AV1 encoded data?

Secondly, I don't understand why rqt as image subscriber declares a parameter in the `ffmpeg` namespace which points to an encoder (`librav1e`) rather than providing the option to set the decoder for av1.

berndpfrommer, 2025-05-31:
1) I'm not an expert on this codec business so I wanted to make sure that on the decoder side there is as much information as possible about the encoder, in it *does* matter for the decoding. You can always map it to the decoder that matches.
2) For the decoder map, the encoders are the keys, the decoders are the values. The subscriber does a lookup, sees "librav1e"  as encoder, and then looks up image_raw.ffmpeg.map.librav1e as a parameter to find the decoder. 
So in your case:
```
ros2 param set /rqt_gui_cpp_node_644409 image_raw.ffmpeg.map.librav1e av1
```
should map librav1e  to the av1 decoder.
Except that it doesn't, because you get this undeclared parameter error. What I don't understand in the first place is that you even see a parameter ``image_raw.ffmpeg.mpa.librav1e`` despite it not being in the [default map](https://github.com/ros-misc-utilities/ffmpeg_encoder_decoder/blob/f044c8d7ab497ddb0cecdc96f074db8d01e9eab0/src/decoder.cpp#L27C1-L32C29). Where does the declaration even come from? Truth be told I may have never even tried to map anything that was not in the default map.

christianrauch, 2025-05-31:
> What I don't understand in the first place is that you even see a parameter `image_raw.ffmpeg.mpa.librav1e` despite it not being in the [default [...]

That only seems to happen when I subscribe directly to the image transport topic (i.e. `/image_raw/ffmpeg`) in rqt. With `image_raw.ffmpeg.encoding:=libaom-av1` I am also seeing `image_raw.ffmpeg.map.libaom-av1` as rqt parameter.

Well, with the default parameters, (`ros2 run v4l2_camera v4l2_camera_node --ros-args -p video_device:=/dev/video2`, no `encoding`) and using the `image_view` for simplicity:
```sh
ros2 run image_view image_view --ros-args -r image:=image_raw -p image_transport:=ffmpeg
```
I indeed see only these defaults:
```
$ ros2 param list /image_view_node
  image_raw.ffmpeg.map.h264_nvenc
  image_raw.ffmpeg.map.h264_nvmpi
  image_raw.ffmpeg.map.h264_vaapi
  image_raw.ffmpeg.map.hevc_nvenc
  image_raw.ffmpeg.map.libx264
  image_raw.ffmpeg.map.measure_performance
```

But with `image_raw.ffmpeg.encoding:=librav1e`, `ros2 param list /image_view_node` will also have `image_raw.ffmpeg.map.librav1e`, which is empty.

> 2\. For the decoder map, the encoders are the keys, the decoders are the values. The subscriber does a lookup, sees "librav1e"  as encoder, and then looks up image_raw.ffmpeg.map.librav1e as a parameter to find the decoder.

With `ros2 param set <node> image_raw.ffmpeg.map.librav1e av1`, I get:
```
[INFO] [1748727331.335997908] [Decoder]: decoding with av1
[av1 @ 0x5d6b0ecd7cc0] Your platform doesn't support hardware accelerated AV1 decoding.
[av1 @ 0x5d6b0ecd7cc0] Failed to get pixel format.
[WARN] [1748727331.343504649] [Decoder]: send_packet failed for pts: 142
g[av1 @ 0x5d6b0ecd7cc0] Missing Sequence Header.
[WARN] [1748727332.045803080] [Decoder]: send_packet failed for pts: 143
g[av1 @ 0x5d6b0ecd7cc0] Your platform doesn't support hardware accelerated AV1 decoding.
[av1 @ 0x5d6b0ecd7cc0] Failed to get pixel format.
```
Which is probably because I do not have a hardware decoder.

But for me, `ros2 param set /image_view_node image_raw.ffmpeg.map.librav1e libdav1d` worked.

> 1. I'm not an expert on this codec business so I wanted to make sure that on the decoder side there is as much information as possible about the encoder, in it _does_ matter for the decoding. You can always map it to the decoder that matches.

Well, I am also not an expert in this as I just started to experiment with this. I imagined that the raw `data` in `FFMPEGPacket` would contain the raw bits of a streamed video file, whose codec is identified by `encoding`. That you have to map the decoder to the encoder is quite confusing, and I am essentially wondering if this is necessary at all. If I download a video file in AV1 codec, e.g. from https://test-videos.co.uk/sintel/mp4-av1, then does VLC need to know which software library created that file?

berndpfrommer, 2025-06-01:
> Well, I am also not an expert in this as I just started to experiment with this. I imagined that the raw data in FFMPEGPacket would contain the raw bits of a streamed video file, whose codec is [...]
You are probably right that av1 or hevc or 264 etc are sufficiently standardized but I just did't want to take my chances. Most users have been able to deal with it so far.

berndpfrommer, 2025-06-10:
Were you able to get everything to work? If yes please close this issue. Thanks!

christianrauch, 2025-06-11:
Yes. I am able to set this up correctly now. But to simplify the setup, I would still consider using the codec name as identified instead of the specific encoder.

closed 2025-06-11 by christianrauch