ros-misc-utilities/ffmpeg_image_transport #48: `ros2 param dump <node>` fails for nodes using ffmpeg_image_transport v3.0.0

Tuntenfisch, 2025-08-22:
I noticed that on v3.0.0 of ffmpeg_image_transport `ros2 param dump <node>` fails under certain conditions and after some time I was able to come up with an MRE.

## MRE

### Context:

- ROS2 Humble (Ubuntu 22.04)
- ffmpeg version 7:4.4.2-0ubuntu0.22.04.1
- ffmpeg_image_transport installed from source as per the [documentation](https://github.com/ros-misc-utilities/.github/blob/master/docs/build_ros_repository.md)
- example.launch.py
  ```python
  import tempfile
  
  import launch
  from launch_ros.actions import Node
  from PIL import Image
  
  
  def generate_launch_description() -> launch.LaunchDescription:
      # Create a dummy image so we have something to publish.
      img = Image.new("RGB", (320, 320), color="red")
  
      with tempfile.NamedTemporaryFile(suffix=".png", delete=False) as temp_file:
          img.save(temp_file, format="PNG")
          temp_file_path = temp_file.name
  
      return launch.LaunchDescription(
          [
              # We need a node that actually publishes something for
              # the republisher to fully initialize the ffmpeg plugin.
              Node(
                  package="image_publisher",
                  executable="image_publisher_node",
                  arguments=[temp_file_path],
                  parameters=[
                      # Ensure the encoder is actually available on your system.
                      {"image_raw.ffmpeg.encoder": "h264_nvenc"}
                  ],
              ),
              Node(
                  package="image_transport",
                  executable="republish",
                  name="republisher",
                  arguments=[
                      "ffmpeg",
                      "in/ffmpeg:=/image_raw/ffmpeg",
                      "raw",
                      "out:=/raw_topic",
                  ],
                  output="screen",
              ),
          ]
      )
  ```

### Steps

1. Open a terminal and launch the file with `ros2 launch <path_to_example_launch_py>`.
2. In a separate terminal, dump the parameters of the republisher node with `ros2 param dump /republisher`

### Expected Outcome

The parameters should be listed.

### Actual Outcome

```bash
user@pc:/workspaces$ ros2 param dump /republisher 
Traceback (most recent call last):
  File "/opt/ros/humble/bin/ros2", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.18.12', 'console_scripts', 'ros2')())
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2cli/cli.py", line 91, in main
    rc = extension.main(parser=parser, args=args)
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2param/command/param.py", line 39, in main
    return extension.main(args=args)
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2param/verb/dump.py", line 119, in main
    self.insert_dict(
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2param/verb/dump.py", line 77, in insert_dict
    self.insert_dict(dictionary[split[0]], split[1], value)
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2param/verb/dump.py", line 77, in insert_dict
    self.insert_dict(dictionary[split[0]], split[1], value)
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2param/verb/dump.py", line 77, in insert_dict
    self.insert_dict(dictionary[split[0]], split[1], value)
  [Previous line repeated 1 more time]
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2param/verb/dump.py", line 79, in insert_dict
    dictionary[key] = value
TypeError: 'str' object does not support item assignment
```

berndpfrommer, 2025-08-22:
I have observed the same thing and don't know how to debug it.
And what is an MRE? I have only heard this acronym as Meals Ready to Eat.

berndpfrommer, 2025-08-22:
I believe ``ros2 param list`` actually works correctly, it's just the dump command that croaks.

Tuntenfisch, 2025-08-22:
I created a simple python node that declares the parameters declared by the ffmpeg plugin and there `ros2 param dump` fails with the same error. Seems like it's an issue with `dump.py`. But I don't know if we would get a fix for ROS2 Humble...

```python
from rclpy.node import Node

class ExampleNode(Node):
    def __init__(self):
        super().__init__("example_node")

        self.declare_parameter("in.ffmpeg.decoder_av_options", "")
        self.declare_parameter("in.ffmpeg.decoder_measure_performance", False)
        self.declare_parameter("in.ffmpeg.decoders.h264", "")
        self.declare_parameter("in.ffmpeg.decoders.h264.yuv420p", "")
        self.declare_parameter("in.ffmpeg.decoders.h264.yuv420p.bgr8", "")
        self.declare_parameter("in.ffmpeg.decoders.h264.yuv420p.bgr8.bgr8", "")

if __name__ == "__main__":
    import rclpy

    rclpy.init()
    node = ExampleNode()
    rclpy.spin(node)
    node.destroy_node()
    rclpy.shutdown()
```

Perhaps one could adjust the way ffmpeg plugin declares it's parameters to avoid the bug in `dump.py`.

Tuntenfisch, 2025-08-22:
Sorry for the spam. But I narrowed it down to this:

```python
from rclpy.node import Node

class ExampleNode(Node):
    def __init__(self):
        super().__init__("example_node")

        self.declare_parameter("in.ffmpeg.decoders.h264.yuv420p", "")
        self.declare_parameter("in.ffmpeg.decoders.h264.yuv420p.bgr8", "")

if __name__ == "__main__":
    import rclpy

    rclpy.init()
    node = ExampleNode()
    rclpy.spin(node)
    node.destroy_node()
    rclpy.shutdown()
```

So seems like the `dumpy.py` takes issue if one parameter name is nested inside another like it is the case here?

berndpfrommer, 2025-08-22:
I thought "." was the correct/recommended separator to use for parameter hierarchies?