ros-misc-utilities/ffmpeg_image_transport #52: Misconfiguration for "Jazzy and later (formerly API_V3)"

roncapat, 2025-09-03:
First of all, thanks for the development of this package!

I just want to report that in your CMake file, you have:
```cmake
if(${image_transport_VERSION} VERSION_GREATER_EQUAL "5.1.7")
  # Jazzy and later (formerly API_V3)
  add_definitions(-DIMAGE_TRANSPORT_USE_PUBLISHER_T)
  add_definitions(-DIMAGE_TRANSPORT_NEEDS_PUBLISHEROPTIONS)
  # add_definitions(-DIMAGE_TRANSPORT_API_V3)
endif()
```
but this fails the compilation if version `5.1.4` is present instead.
This may go back as `5.0.0` _maybe_, but `5.1.4` is the only one I tested.

berndpfrommer, 2025-09-03:
This could well be. I probably just looked at what the current version of the package was for the distro I was targeting.
What distro are you running on?

roncapat, 2025-09-03:
A custom Jazzy-ish ROS distro from source - something in between Jazzy patch release 4 and patch release 6 - plus some (here completely unrelated) patches of my own.

berndpfrommer, 2025-09-03:
@roncapat Thanks for reporting this issue. Closed with PR #53.

closed 2025-09-03 by berndpfrommer